probatio/probatio #2: Improve documentation and organization

DocSunset, 2021-03-24:
The layout of repositories and documentation is currently fragmented and incomplete. This makes it challenging for newcomers to make sense of the current state of the system, how everything fits together, how to replicate the system, and how to participate in the project. Among other efforts, such as the [BOM calculator](https://github.com/probatio/probatio-calculator), significant improvements could be made to the documentation and organization of the source files for the project.

DocSunset, 2021-03-24:
Since this repository (probatio/probatio) is already the main container for the existing project documentation (in the wiki), I propose that we also use it as a single point of contact and organizing container for the myriad other repositories in the project using submodules. This will allow developers to see a high-level overview of how the repositories relate to each other and which are most relevant, as well as retaining the advantages of having numerous separate repos (e.g. individual version control).

jtragtenberg, 2021-03-25:
Yes, that would help a lot. How can we migrate the issues to this new structure? Maybe only add new issues there?

DocSunset, 2021-03-25:
I assumed we would simply stop making issues in other repos and only make them here moving forward; I think it's even possible to disable issues as a repo feature, which we might do in the future if we do migrate to using one repo for all issues.

It is possible to migrate issues one at a time to a different repo; the button is labelled "Transfer Issue" on my screen and appears on the bottom right of each issue page on desktop with a -> arrow icon. There aren't that many issues total right now, so it wouldn't be too arduous to transfer them all to one repo, but I'm not sure it's worth it? Maybe it is? Really not sure.

jtragtenberg, 2021-03-25:
Sounds good but not sure either. Lets discuss the pors/cons in a next meeting

filipecalegario, 2021-03-26:
> Yes, that would help a lot. How can we migrate the issues to this new structure? Maybe only add new issues there?

I think we could keep the specific issues in the specific repositories because we can have the issues' overview in the project [Projeto Rumos](https://github.com/orgs/probatio/projects/2.) 

What do you think? @jtragtenberg @DocSunset

filipecalegario, 2021-03-26:
> I assumed we would simply stop making issues in other repos and only make them here moving forward; I think it's even possible to disable issues as a repo feature, which we might do in the future if [...]

Maybe it is better to modularize the issues. We could have contributors that, for example, only deal with 3dprinting, or others that only deal with firmware. I think it would be complicated to manage all the issues in just one place. 

We can organize the overview of issues in [Projeto Rumos](https://github.com/orgs/probatio/projects/2).

DocSunset, 2021-03-26:
> Maybe it is better to modularize the issues. 

I replied in #3.