probatio/probatio #3: Discuss github issue workflow

DocSunset, 2021-03-24:
Hi all,

I'm adding a bunch of issues right now and wanted to discuss my intent here. 

First of all, anyone should feel welcome to edit, move, or outright delete any of the issues I'm opening if they have strong feelings about where or whether they should exist.

Second, I've tried to put issues in the relevant repo (3D printing issues in the printing repo, firmware issues in the firmware repo, etc). Where the issue would seemingly require changes across multiple repos, I've put them on this repo.

I would like to suggest that we consider making all issues in this one repository moving forward and use labels to identify what aspect(s) of the project they relate to. It's kind of a pain trying to decide which repo to put any given idea in, having issues all spread out also makes it difficult to search issues, and it would probably be more convenient moving forward to have them all in one place. Right now the Probatio Rumos project is acting as a top level container for all issues, but I would argue the issues page is better suited to containing and searching issues, and that projects should be used to organize projects e.g. sprints and large multi-part enhancements.

What do you think?

DocSunset, 2021-03-24:
Having all issues in one place would also make it easy to keep abreast of ongoing work by subscribing to one repo.

DocSunset, 2021-03-24:
Only one column (category?) from a project, or one repo, can be assigned to an issue at a time, which make it especially hard to organize issues that cut across aspects of the project, requiring code, 3d printing, and pcb changes, for example. Labels are better suited to this sort of organization that either projects or repos (at least for issues)

DocSunset, 2021-03-24:
Apparently github has a "Discussions" feature that would also be handy, but it has to be enabled in the repo settings, which I can't access.

DocSunset, 2021-03-24:
There also doesn't seem to be a way to automatically add all issues created to a project, which further makes me believe a project page is not an ideal container for all issues across repos.

I guess you can always do a search within the probatio organization to find all issues, but this still seems less natural than using an issue page.

DocSunset, 2021-03-25:
Projects do allow you to arbitrarily order cards in columns though, which is a very handy feature for prioritizing compared to issues on the issue page that can [only be sorted by metadata](https://docs.github.com/en/github/managing-your-work-on-github/sorting-issues-and-pull-requests)

jtragtenberg, 2021-03-25:
I believe the KANBAN projects are great to organize specific time-bound projects like the Rumos (financed by a bank for one year), but I agree they are also good to manage the issues in the repository/ies as a whole.

I like your proposition to centralize all the issues in one repository. I believe once you start to centralize all the content in the probatio/probatio with submodules we could start issuing there.

DocSunset, 2021-03-25:
Ah, I misunderstood the purpose of that project. I’ve been adding all issues to it, but maybe they’re not all within its scope.

DocSunset, 2021-03-25:
> I like your proposition to centralize all the issues in one repository.

I'm going to interpret this as permission to go ahead and start making all future issues in this repository starting now... We can always migrate them out later if we decide we'd rather not centralize everything.

jtragtenberg, 2021-03-25:
Do you agree @filipecalegario ?

DocSunset, 2021-03-25:
I'm also happy to transfer issues from other repositories into this one if that's something we want to do.

filipecalegario, 2021-03-26:
> Do you agree @filipecalegario ?

I've raised some questions in this [reply](https://github.com/probatio/probatio/issues/2#issuecomment-807920179).

I think we have to modularize the issues to specific projects and try to find a way to concentrate them in one place for us that are dealing with all the aspects of Probatio.

We have to consider contributors to specific projects that won't deal with other projects' issues. For example, a developer that is coding improvements to the firmware is not necessarily interested in issues related to 3d-printing.

I think that Probatio is many projects into one and as soon as we break the modules and their safe isolation, it could be difficult to allow specific contributions due to high complexity.

filipecalegario, 2021-03-26:
> Ah, I misunderstood the purpose of that project. I’ve been adding all issues to it, but maybe they’re not all within its scope.

I don't think it is a problem @DocSunset since we're put so many things in it that I'm considering it as a backlog.

@jtragtenberg I think the project [Probatio Rumos](https://github.com/orgs/probatio/projects/2) can be repurposed as a more general project that covers things beyond our current development. Do you agree?

filipecalegario, 2021-03-26:
We could use the following steps:

- Bugs and enhancements -> create issues in the specific repository.
- Ideas of new blocks, bases, and supports -> create issues in probatio/probatio (here)
- General discussions or issues that are hard to fit in other repositories -> create issues in probatio/probatio (here)

We can periodically transfer misfitted issues to make everything clean.

What do you think? @jtragtenberg @DocSunset

filipecalegario, 2021-03-26:
For us, as we're dealing with different aspects of the project, we should find a way to automatically aggregate the issues from different repositories in the same fashion as the submodules in probatio/probatio, which is a perfect solution to maintain modularity and encapsulation.

ChristianFrisson, 2021-03-26:
> For us, as we're dealing with different aspects of the project, we should find a way to automatically aggregate the issues from different repositories in the same fashion as the submodules in probatio/probatio, which is a perfect solution to maintain modularity and encapsulation.

@filipecalegario It is possible to configure automation for project boards (https://docs.github.com/en/github/managing-your-work-on-github/configuring-automation-for-project-boards) I suspect only by owners, including for auto-adding issues to a project (https://github.community/t/new-issues-not-being-auto-added-to-project-board/904/1).

DocSunset, 2021-03-26:
It seems we have identified the following requirements for our workflow:

1) all issues are accessible in a central location for developers working across the whole project
2) subsets of issues related to specific parts of the project (e.g. firmware, 3D modelling, electrical design, specific blocks) are easily accessible for users working on parts of the project, without them having to dig through the entire backlog of all issues (facilitate specific contributions without having to navigate high complexity)
3) minimize manual actions required by developers to follow the system

And we have roughly speaking two proposals.

## Single repo for all issues, labels for project aspects

Workflow:
- add all issues to probatio/probatio
- add labels when creating issue to identify what parts of the project are implicated (e.g. firmware, 3D modelling)
- to view all issues, go to the issue page for probatio/probatio
- to view a subset of issue related to one part of the project, search for the appropriate label

Pros:
- meets requirements
- minimum amount of github automation required
- makes cross referencing issues easier since typing "#" provides text-searchable list of all issues
- easier to do searches across all issues, e.g. if you think an issue might already exist
- minimum decision paralysis; just add it to probatio/probatio and add as many labels as you think it might need

Cons:
- developers can forget to add project part labels, may require maintenance to ensure all issues are appropriately labelled (i.e. people can make mistakes)

## Many repos, single github-project-based backlog

Workflow:
- add issues to the most appropriate repo
- *add cross-cutting issues to probatio/probatio (i.e. use workflow 1 as described above?!)*
- all issues are automatically added to an organization-wide project using github automation
- to view all issues, go to the backlog project
- to view a subset of issues, go to the appropriate repo *AND check in probatio/probatio*

Pros:
- meets requirements

Cons:
- requires maintaining github automation
- harder to cross-reference issues across repos (but still possible)
- developers have to check in two places if working on only part of a project due to cross-cutting issues in probatio/probatio
- searching for existing issues requires using non-standard issue search procedures (probably the github search bar "in this organization") or checking in multiple repositories
- people can obvious still make mistakes

---

I think it's pretty obvious from my very biased analysis that the single-repo-for-all-issues strategy, which (important!) uses labels to identify the parts of the project implicated by an issue, is a better approach. The fact that cross-cutting issues are sure to keep coming up makes an approach using multiple repositories less efficient, because you can't add multiple repositories to an issue. So you necessarily end up using the first system as a subset of the second system, which itself arguably includes an additional system that needs maintenance due to the need for automation. It seems clear to me that one system is better than three systems.

DocSunset, 2021-03-26:
I took the liberty of adding appropriate labels to the issues already in this repo. You can try searching for e.g. issues related to firmware only on the issue page now to see what you think of this approach compared to having issues in separate repos. (i.e. we are now using system 2 (Many repos, single github-project-based backlog) as described above. Your firmware related issue might be here, or in the firmware repo.)

DocSunset, 2021-03-26:
Even with all issues in a main backlog project you can't easily find everything related to firmware because, unless I'm mistaken, there is no way to filter cards with a logical OR search query. I can't search for "label:firmware OR repo:probatio/probatio-firmware", I have to search one and then the other.

DocSunset, 2021-03-26:
You also can't use the github search bar to find all issues related to firmware; a search like "org:probatio AND (is:issue OR repo:probatio/probatio-firmware)" doesn't work. 

You seriously just have to check in two places or make at least two searches.

jtragtenberg, 2021-04-06:
Sorry for the late reply, I received and e-mail with all the issues posted here but a week later, when I could answer them, I tried coming here to github and I just couldn't find this issue. I looked in the Rumos Probatio Project, in the probatio-docs, in many other specific repos and forgot this probatio/probatio was already being used.
This difficulty is a good point for @DocSunset's argument. I believe using a centralizing project wouldn't be as good because in the project there are many dispersed threads. One only repo for the issues with the appropriate tagging would be in my humble opinion the best solution.

jtragtenberg, 2021-04-06:
> I don't think it is a problem @DocSunset since we're put so many things in it that I'm considering it as a backlog.
>
> @jtragtenberg I think the project [Probatio [...]

I believe it is good to have KANBAN projects to manage specific projects, like the Rumos or the one with McGill. It is good to organize specific goals and improvements in each project's focus.

jtragtenberg, 2021-04-06:
> I've raised some questions in this [reply](https://github.com/probatio/probatio/issues/2#issuecomment-807920179).
>
> I think we have to modularize the issues to specific projects and try to find a way [...]

@filipecalegario, do your think adding appropriate tags would solve this issue for contributors to specific projects?

filipecalegario, 2021-04-07:
@jtragtenberg  @DocSunset 
Let's try this approach then and gather all issues in probatio/probatio and organize them in tags.

I think we should make the tags closer to the name of the repositories.
For example, the tag `structure`, at a first glance, is not directly linked to `3dprinting`.
How should we define the tags to make these links more obvious? 

I will transfer the issues of the specific repositories to here, OK?
In the meantime, I'll take the opportunity and review the issues.

filipecalegario, 2021-04-07:
I've just reread and the tag is not only `structure` but `physical structure`. In that case, I think it's fine. =D