probatio/probatio #10: Collect printing documentation for each model in an easy to reference place

DocSunset, 2021-03-25:
I agree we should think of a way to suggest which ones need supports. Maybe a tag at the end on the file name like probatio_block_top_dual_pistons_cap(use_supports).stl or just a “*”. Not the most elegant solution, a table would also be good but I think the file name communicates faster For someone in the printing labor :)

It would be nice to have a table showing 
- a rendered image of each 3d model in the direction it should be printed, 
- a link to the fusion file, 
- which models should be printed together and a visual representation of how the connect, 
- maybe a picture of it with the sensor used, 
- if it need to be printed with supports

_Originally posted by @jtragtenberg in https://github.com/probatio/probatio-docs/issues/5#issuecomment-806835379_

DocSunset, 2021-03-25:
I like the idea of encoding information directly in the filenames of the STLs in a way that's reasonably easy to parse just by reading while you're printing things. Maybe the files could be named automatically by parsing a CSV or JSON file or something that could also be used to generate a pretty table.

DocSunset, 2021-03-25:
Would it be better to have a table that's oriented towards individual models (e.g. with a row for each 3D model), or oriented around individual assemblies that one might build (with a row for each block, each base/support, etc)? Maybe both?

jtragtenberg, 2021-03-25:
I believe the table should have rows and subrows maybe for objects that need more than one stl. 
I like this scripting approach for the file names and the table if it doesn’t add an overhead too big to us. We should definitely standardize the file names. Now some of them use “ - “ and some use “ _ “ ...

DocSunset, 2021-03-25:
> I like this scripting approach for the file names and the table if it doesn’t add an overhead too big to us. 

This could possibly be incorporated into the calculator web app. It may not be necessary or desirable for the STLs in the repo to have long verbose informative names. 

Though maybe it would be nice? What is easier to maintain though?

It seems likely that the calculator web app will end up being the main authority on how all the bits in the various repositories come together, so it may be the most natural place to deal with making this kind of documentation in a way that can be maintained (ideally) semi-automatically.

jtragtenberg, 2021-03-25:
I like the verbose names because it helps to find the right stl when printing. They organize well in alphabetical ordering. 
We could rethink the names to match the table in the logic of grouping objects with many stls. The 1x3 base is a bit confusing for example since the neck is labeled as a support and the necessary block bottom isnt considered part of the 1x3 base. Should I add a new issue to that?

DocSunset, 2021-03-25:
> The 1x3 base is a bit confusing for example since the neck is labeled as a support and the necessary block bottom isnt considered part of the 1x3 base. Should I add a new issue to that?

I say the more issues the merrier. It's a great way to keep track of ideas and ways to improve that are otherwise often lost to the nebulous flow of time.

jtragtenberg, 2021-03-25:
I just realized I had already raised an issue about it:
https://github.com/probatio/probatio-3dprinting/issues/3 (as you said before, not being able to link an issue to another repo is a good point to keep them all in the same place)

jtragtenberg, 2021-03-25:
probatio/probatio-3dprinting#3 test