probatio/probatio #11: Standardize file naming conventions (and other matters of style and consistency)

DocSunset, 2021-03-25:
We should definitely standardize the file names. Now some of them use “ - “ and some use “ _ “ ...

_Originally posted by @jtragtenberg in https://github.com/probatio/probatio/issues/10#issuecomment-806922037_

For code in particular, we may wish to adopt an existing style guide if we can agree on one, ideally something with automated tooling for verifying adherence. For file names I’m not sure there are such guides?

filipecalegario, 2021-03-26:
Let's use “ _ “ as the standard for file names. I've already fixed the naming conventions of STLs in this [commit](https://github.com/probatio/probatio-3dprinting/commit/09c0a0b3661c8b360b1f41f3a897b26f3df081f2#diff-b840cc1780c860f1c3893bf447249194eca14c5b7c0d9855832c04412b433c4b)

 We should check other files in other repositories.

DocSunset, 2021-03-26:
Hold up.

The repositories all already use dashes as a word separator.

My lizard brain craves consistency. And laziness. It's a pain to change repository names. It's easy to change filenames.

Maybe we should stick to using dashes everywhere?

We could also then use underscore as a kind of field separator. So the STLs could be named with a convention like:

`assembly-name_part-name_version-number.stl`

DocSunset, 2021-03-26:
I've started a style guide in the wiki: https://github.com/probatio/probatio/wiki/Style-Guide

I'll try to keep it up to date as we discuss and make decisions