probatio/probatio #12: Discussion: Switch to openSCAD or other open modelling software?

DocSunset, 2021-03-25:
Should we switch to an open source 3D modeller such as openSCAD, freeCAD, solvespace, or another?

Fusion360 pros:

- excellent user interface and user experience
- very powerful!
- familiar to several of us
- existing models already finished in fusion360

Fusion360 cons:

- proprietary software
- full power is behind a pay wall
- mild vendor lock in
- uncertain compatibility with git version control and collaboration workflow
- ... proprietary software... >:(

Open source alternatives pros:

- free, libre, open source, human rights respecting
- generally good git workflow integration
- documentation generally better than fusion360 (imo)
- easier for newcomers to participate in development without having to pay money or make an account with a private corporation
- possibly better integration with other open source tools, scripting, etc?

Open source alternatives cons:

- generally less polished user experience
- generally not as powerful as easily (may take more effort to achieve similar results)
- may require us to learn new tools
- would have to re-create many existing models

DocSunset, 2021-03-25:
I will be exploring the git-compatibility of fusion360's exported design files. If things integrate with git more smoothly than I am expecting, I will adjust the pro-con list in the first post.

jtragtenberg, 2021-03-25:
I am now committing some f3d and f3z files from Fusion 360 I designed. This f3z can include external components, like 3d modes of the buttons, potentiometers that we can download from the internet in step files. Is it easy to do that in the open source alternatives? They are really handy to communicate to the maker what components to put use with that 3d model

DocSunset, 2021-03-25:
Yeah, handling multi-part assemblies is something fusion360 excels at. I believe similar functionality is possible in freeCAD using add-ons, but I don't think any open source packages can handle this sort of thing out of the box. I'll add mention of this to the pro-con list above.

DocSunset, 2021-03-27:
Added some more analysis of specific open source packages to the first comment.