probatio/probatio #13: Help in git workflow

jtragtenberg, 2021-03-25:
Since I'm new to git workflow, I don't know how to proceed. I'm afraid to mess up with the repo so I'm asking for help:

I didn't have the latest version of the probatio/probatio-3d-printing repository and I created and renamed some files in it. I can't push before updating it, but I don't know if by git pulling it I will lose anything.
I got this error message: 

```
 ! [rejected]        master -> master (fetch first)
error: failed to push some refs to 'https://github.com/probatio/probatio-3dprinting.git'
hint: Updates were rejected because the remote contains work that you do
hint: not have locally. This is usually caused by another repository pushing
hint: to the same ref. You may want to first integrate the remote changes
hint: (e.g., 'git pull ...') before pushing again.
hint: See the 'Note about fast-forwards' in 'git push --help' for details.
```

What should I do before pushing my commit?

jtragtenberg, 2021-03-25:
I tried to do a "git pull" and I got this message. I believe some file changed by someone else I had also changed, probably a file name:
```
Merge branch 'master' of https://github.com/probatio/probatio-3dprinting
# Please enter a commit message to explain why this merge is necessary,
# especially if it merges an updated upstream into a topic branch.
#
# Lines starting with '#' will be ignored, and an empty message aborts
# the commit.
```
What should I do here?

DocSunset, 2021-03-25:
There are probably many ways to deal with this. I think (if I understand the situation) what I would do is 

- `git checkout -b filenames`, then commit your local changes to the filenames branch. 
- Then you can `git checkout master` and pull the remote changes, `git checkout filenames' to switch back to your feature branch, and `git merge master` to merge the remote changes into your feature branch. 
- Then you can either `git checkout master` and `git merge filenames` to merge your changes locally before `git push`ing changes upstream, 
- or you can `git checkout filenames`, `git push -u origin filenames' to push your local changes to a feature branch upstream, and then you can open a pull request on github and ask for a review before your changes get merged to master.

DocSunset, 2021-03-25:
That might be a bit convoluted though...

DocSunset, 2021-03-25:
You might be able to just commit your local changes, and then do `git pull`; it will complain that it can't automatically merge, and you will have to manually deal with the conflicts and `git commit` to finish the merge yourself?

DocSunset, 2021-03-25:
The text box in your second comment is just composing a commit message, specifically one for your merge from upstream. I think you can just save and quit your text editor and you'll be done a merge.

DocSunset, 2021-03-25:
You address your original concern directly, 

>I don't know if by git pulling it I will lose anything I just did

As long as you had already run `git add` on your local changes, it's very difficult to lose them. If you already committed them, it's almost impossible to lose them without deliberately deleting things.

jtragtenberg, 2021-03-25:
Great, thank you. That is what I wanted to know. I will try next time to create a new branch. So the second best workflow when this happens without using branches it to:
```
git add . 
git commit -m "..." 
git pull
git push

```
?

jtragtenberg, 2021-03-25:
or should I always first do a "git pull" before anything?

DocSunset, 2021-03-25:
I would say it’s always best to pull before you start making changes, but once you’ve made changes the commands you describe are exactly right, assuming you are working on a branch that is already set up with an upstream remote. 

Pull
Work
Add
Commit
Push

If push fails because of upstream changes:
Pull
(Possibly manually complete the merge)
Push

It would be the same if you were on a feature branch collaborating with someone as if you are directly pushing and pulling to the main branch.

DocSunset, 2021-03-25:
Obviously if you’re working on your own local feature branch by yourself it would be a little bit different, e.g.

```sh
git checkout -b featurename 
# work on your feature
git add changed.file fileglob.* etc.txt
git commit -m ‘...’
# repeat from “work on your feature” until done working and ready to push upstream
git push -u origin featurename
```
And then you can open a pull request on GitHub, and switch to the workflow as described for a branch with remote upstream if you need to collaborate with others to make it ready to merge.

DocSunset, 2021-03-27:
Closing this issue for now. Please reopen if you have similar issues again; otherwise, for novel git/github challenges we can open a new issue.

closed 2021-03-27 by DocSunset