probatio/probatio #8: Improve design for manufacturing / design for assembly

DocSunset, 2021-03-25:
Assembling a single block currently consists of (roughly) these steps:

- print block base
- print block top
- clean rough edges of base and top as required based on the quality and calibration of your printer
- solder pins onto two pogo pcbs
- connect (i.e. solder) pogo pcbs with point-to-point fly wires
- connect pogo subassembly to arduino board with point-to-point fly wires
- connect sensors to arduino board with point-to-point fly wires
-

DocSunset, 2021-03-25:
@jtragtenberg @filipecalegario please edit the above description of the current assembly process in case it doesn't reflect reality.

jtragtenberg, 2021-03-25:
Yes, this seems right to me. Do fly wires mean regular flexible wires?
I will measure how long this process takes aproximately.

I am testing also do do some sanding on the exterior of the block for it to have a smoother insert/removal from the base. While gluing the pogo-pcbs, this step can only be done before gluing them in place.

I completely agree that we should reduce the gluing and soldering. One thing to keep in mind however is to maintain the flexibility of the setup so it is easy for people to create new blocks without the need of designing And ordering a new pcb. I believe at this moment flexible wires are the most flexible solution for future hacks I can think of.
But for structural elements such as pogo-pcbs we should definitely change for a faster to assemble structure. It would amazing if everything snap-fitted to block bottom quickly.

jtragtenberg, 2021-03-25:
I would also add the suggestion here to use the base-pcbs as jigs to glue the magnets. This is a useful tip, since it is a nightmare to glue such strong magnets together.

DocSunset, 2021-03-25:
> I would also add the suggestion here to use the base-pcbs as jigs to glue the magnets. This is a useful tip, since it is a nightmare to glue such strong magnets together.

This might be made simpler by using stacks of thin magnets as I suggested [here](https://github.com/probatio/probatio-3dprinting/issues/11#issuecomment-806842882)

DocSunset, 2021-03-25:
> Yes, this seems right to me. Do fly wires mean regular flexible wires?

Yes, exactly.

> I completely agree that we should reduce the gluing and soldering. One thing to keep in mind however is to maintain the flexibility of the setup so it is easy for people to create new blocks without [...]

I completely agree. Having an optional PCB to speed up assembly of existing block designs doesn't necessarily prevent anyone from still hacking together a new block with bits of wire though. The two can coexist as long as the interface between the block-specific components and the block-generic/common parts is really strict and clearly defined.

> But for structural elements such as pogo-pcbs we should definitely change for a faster to assemble structure. It would amazing if everything snap-fitted to block bottom quickly.

Yeah, these are parts that every block has in common, so it should ideally be really fast and easy to assemble them.