probatio/probatio #15: Updates in repositories and submodules related to it

jtragtenberg, 2021-04-06:
Hello, can you please explain to me how the submodules work? Why are they linked to a specific commit of another repository?

filipecalegario, 2021-04-06:
I think we should commit to the specific repositories and deal with the integration automatically in probatio/probatio later.

DocSunset, 2021-04-06:
> Why are they linked to a specific commit of another repository?

That is the expected behaviour. Basically, the submodule just points to a specific commit of another repository, which is intended to allow that external repo to be pulled into the host repo as a dependency. Updating to a newer commit is like upgrading a package for the host repo, and involves making a commit from the host repo. 

There are actually a lot of gotchas with submodules, so it's best to continue to work using the many seperate repos. The main probatio/probatio repo can then be used for the sole purpose of keeping a record of what versions of the many separate repos are known to work together, and as a place where you can view the whole project at a high level. It's not really ergonomic to use it as a workplace, from what I can tell (which came as a disappointment to me also), although it may be possible?

Here's an article I found helpful: https://medium.com/@porteneuve/mastering-git-submodules-34c65e940407

filipecalegario, 2021-04-06:
To highlight the important point, probatio/probatio is the aggregator of the different repositories and should not be used as a workplace. We should contribute to the specific repositories.

jtragtenberg, 2021-04-06:
Ok, I will close this issue then. If someone has more to discuss about it just open it again

closed 2021-04-06 by jtragtenberg