probatio/probatio #29: Create one repository for all PCBs

filipecalegario, 2021-04-07:
Currently, there are many repositories for Probatio's PCBs. 
A better approach would be to have all of the PCBs in just one repository.
It can simplify for contributors to have everything in one place.

DocSunset, 2021-04-07:
Further consolidation seems positive to me. (I would even argue in favor of one repository for everything if anyone wanted to have that discussion.)

jtragtenberg, 2021-04-08:
> Further consolidation seems positive to me. (I would even argue in favor of one repository for everything if anyone wanted to have that discussion.)

Isn’t the probatio/probatio repository exactly that? One repository for everything

DocSunset, 2021-04-08:
Yeah, but it would be even simpler if we could do it without submodules.

filipecalegario, 2021-04-08:
> Yeah, but it would be even simpler if we could do it without submodules.

Let's move the discussion to the specific issue: https://github.com/probatio/probatio/issues/30